bigtreetech/Manta-M4P #3: Manta-M4P with CB1 and BTT HDMI5

Kudach, 2022-09-02:
I have a problem with the activation of the BTT HDMI5 in the Manta-M4P and the CB1.

How do I activate the HDMI port so that I can use the manta board completely?

billru05, 2022-11-16:
Once you load the OS on the CB1 the HDMI should work, that's providing you did not install the dt-blob-disp1-cam1.bin, if you did not load the dt-blob-disp1-cam1.bin and your HDMI is not working you may need to look at your config.sys file in the boot dir

Tbone5114, 2022-12-29:
Did you find a solution?

billru05, 2022-12-29:
> Did you find a solution?
Solution to what or is this reply for ([Kudach](https://github.com/Kudach))

Tbone5114, 2022-12-29:
This was for @Kudach. I'm running into the same issue where my HDMI isn't working.

Mig1880, 2023-10-06:
I found I had to power the hdmi5 via a USB C with hdmi connected. My issue is now I'm only getting 2/3rd of a screen 
[image]

javcab2003, 2024-02-17:
> I have a problem with the activation of the BTT HDMI5 in the Manta-M4P and the CB1.
> 
> How do I activate the HDMI port so that I can use the manta board completely?

Hi, did you find the solution?

I have the same issue, I have the CBI and the EZE3 but I cannot see anything in the display, in the web interface I can see the USB recognised for the HDMI connection but the display "HDMI 5" shows nothing. I tried to reinstall the Klipper screen and checked the log files but there is nothing that is in error status or similar
[image]