bigtreetech/Manta-M4P #4: M4P / M8P CAN support with EBB42

cstrachan88, 2022-09-16:
I would like to use the M4P (with a Raspberry Pi CM4) and the EBB42 tool board. Is this combination possible? I will be running Klipper on the raspberry pi which supports CAN, but there are no shown outputs to connect to the EBB42 board. Would using the U2C as an intermediary suffice?

If not, are there any plans for future M4P style boards with CAN capabilities? If the M4P had the EZ style motor controllers, CAN-FD, and an CM4 socket, it would be perfect.

TURNERO, 2022-10-19:
I've not tried yet, since my board hasn't arrived but I'm planning on using one of the UART ports on the board to drive an SN65HVD230 canbus tranceiver. These instructions are confirmed to work for other MCUs, not just the PICO: https://github.com/rootiest/zippy-klipper_config/blob/master/guides/Guide-pico_can.md.

I'll let you know at the weekend if it works. If my board doesn't arrive soon, I'll try it with my SKR 1.3.

tnap1979, 2022-10-25:
You can use the u2c with ebb42 just fine, as the u2c acts as a USB mcu with can support

dury10, 2023-07-01:
> I've not tried yet, since my board hasn't arrived but I'm planning on using one of the UART ports on the board to drive an SN65HVD230 canbus tranceiver. These instructions are confirmed to work for other MCUs, not just the PICO: https://github.com/rootiest/zippy-klipper_config/blob/master/guides/Guide-pico_can.md.
> 
> I'll let you know at the weekend if it works. If my board doesn't arrive soon, I'll try it with my SKR 1.3.
> 
> It's frustrating they didn't add the CAN chip and port to the board. I think it's where things are going.

did you manage to make it run on can?

Coolbobjoe, 2023-12-18:
> You can use the u2c with ebb42 just fine, as the u2c acts as a USB mcu with can support

Have you figured out how to do this? I cannot for the life of me get a usable uuid off the device and have semi-confirmed I dont have a bad chip by flashing within STM32.