bigtreetech/Manta-M4P #5: RGB ports configuration

acosano, 2022-10-26:
Hi.  Ispend q few days now trying to get the RG porst to work with no sucess. I try diferent  RGBW and still no go. I'musing klipper and the code is I'm using works on any other printer wbut not on the MP4. Any advice what I'm doing wrong.
[neopixel logo_rgb]
pin: PD0 or PD1 no sucess
chain_count: 24
color_order: GRB
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.0

fastslash25, 2022-11-23:
Hey, I was able to use the PD1 RGB port on my m4p.
I use this in my printer.cfg:

[neopixel my_led]
pin: PD1
chain_count: 23

And I use: 
https://github.com/digitalninja-ro/klipper-neopixel/blob/1ec86884bd5b384e4b6a35c39c5a0f6598d76189/led_progress.cfg

acosano, 2022-11-23:
I use the same cfg and it works fine on the Octopus Pro but for some reason  Ican't get it to work on the MP4. Wil lgive it tomorrow another try.

Aizel330, 2022-12-22:
hey, any luck on this? I'm having the same problem and I really can't figure out what I'm doing wrong...

lyonmaster, 2023-01-16:
Any resolve on this issue, i cant get mine to work either.

emersonsc, 2023-02-07:
Same issue. Neither of my RGB ports function

Aizel330, 2023-02-20:
Hey guys I've solved my problem by connecting an addictional vcc and gnd to my leds, apparently the board alone it's not able to drive many leds so it's better if you get your 5/12v elsewhere and just control the leds via the control pin

Coolbobjoe, 2023-12-18:
> Hey guys I've solved my problem by connecting an addictional vcc and gnd to my leds, apparently the board alone it's not able to drive many leds so it's better if you get your 5/12v elsewhere and just control the leds via the control pin

Not sure if this will help anyone down the road, if you are having issues with flickering/ random colors/ neopixels not activating at all, and you are using a separate power source. Try linking the ground of your separate power source not just to the neopixels themselves, but to the ground port on the rgb port itself, in addition. This helps with a floating ground issue that can cause headaches with longer chain lengths!