bigtreetech/Manta-M4P #6: Adxl345 wiring

Freddyb888, 2022-11-02:
How do I wire a adxl345 on the Manta-M4P with cb1?

Aizel330, 2022-11-07:
Hi dear, I managed to wired and use an adxl345 using the spi port on the m4p, I'll attach some images, hopefully they are clear enough
[image]
[image]

then basically it's just matter of connecting these pins together and add this to your printer.cfg

[adxl345]
cs_pin: PD9
spi_bus: spi1
axes_map: y,x,z

the axes_map need to be modified depending on the orientation of your particular adxl345 so some testing may be needed to correct the orientation and make sure that the compensation is on the right axes. if anyone needs more infos I'll be more than happy to help.

Freddyb888, 2022-11-11:
Thanks, but is not possible to connect to the 40 gpio pins that are present on the btt manta m4p with cb1? Thanks again

Aizel330, 2022-11-11:
https://github.com/bigtreetech/CB1/issues/14#issuecomment-1264667543 I think that could help you with that, hopefully.

Freddyb888, 2022-11-11:
Thanks a lot. I think it will work.

closed 2022-11-11 by Freddyb888

Aizel330, 2022-11-12:
no problem, enjoy!

ox006, 2023-03-16:
how can i connect mpu6050 with M8p/cb1