bigtreetech/Manta-M4P #22: ls /dev/serial/by-id/ always returns no file or directory found.

Park8706, 2023-05-30:
I have used both a compiled flash and one downloaded on the github along side the OS and in both cases I get the same error with my USB plugged into my PC or 3d Printer.

yzheka, 2023-06-17:
What does `dmesg` command says? I had similar problem and it appeared that USB was shut down due to EMI produced by my PSU.

Park8706, 2023-06-19:
> What does `dmesg` command says? I had similar problem and it appeared that USB was shut down due to EMI produced by my PSU.

I currently don't have the board as our dev has it atm. The issue for me ended up being I needed to flash the board without the CB1 board plugged into the manta for it to properly take the firmware. After that it worked fine and the serial pulled up.

Coolbobjoe, 2023-12-18:
I have no idea why this worked for me but...
try ls /dev/serial/by-path/

This was the only way I could find a path, not sure exactly where I found this, I had been struggling for over 2 days to get a usable pathway but this for some reason worked!

yzheka, 2023-12-18:
I have noticed that if i installed a 32 bit OS image it appears normal

Pneumanifest, 2023-12-20:
> reason

This is likely due to using a distro that has the known bug that causes this exact issue. 
You can find more information about that here https://klipper.discourse.group/t/debian-bullseye-bug-causing-klipper-to-no-longer-find-the-printer-board/8231

warpwookie, 2024-03-17:
@yzheka where is the 32bit image file can you provide a link?

yzheka, 2024-03-17:
> @yzheka where is the 32bit image file can you provide a link?

I used raspberry pi imager app

warpwookie, 2024-03-20:
> > @yzheka where is the 32bit image file can you provide a link?
> 
> I used raspberry pi imager app

Are you using CB1 or CM4?

yzheka, 2024-03-21:
> > > @yzheka where is the 32bit image file can you provide a link?
> > 
> > 
> > I used raspberry pi imager app
> 
> Are you using CB1 or CM4?

That is strange, now i have no issue with both cm4 and cb1 using latest os image. Both are working fine.