bigtreetech/Manta-M4P #24: How to connect SFS 2.0?

asychev, 2023-09-20:
One cable from SFS 2.0 goes to E0-STOP, this is clear.
Where to connect second?

Pneumanifest, 2023-09-22:
To another available endstop. Like E1-stop. It really depends on your board and setup.

asychev, 2023-09-22:
> To another available endstop. Like E1-stop. It really depends on your board and setup.

What other endstops besides not available XYZ can be used?

Pneumanifest, 2023-09-22:
> > To another available endstop. Like E1-stop. It really depends on your board and setup.
> 
> What other endstops besides not available XYZ can be used? I hope it was clear the question is about M4P because, well, this is M4P repository, isn't it?..

Like E1-STOP, it depends on what board you have to what is available.

asychev, 2023-09-23:
> Like E1-STOP, it depends on what board you have to what is available.

It is some joke or what? We are in M4P repository, so it means we talk about (surprise!) M4P. And it does not have other usable endstops like E1-STOP.

Pneumanifest, 2023-09-23:
> > Like E1-STOP, it depends on what board you have to what is available.
> 
> It is some joke or what? We are in M4P repository, so it means we talk about (surprise!) M4P. And it does not have other usable endstops like E1-STOP.

My sincere apologies. I was in another repo and talking with several others I did notice the repo and assumed it was for the SFS2. For the M4P you might be able to use the probe pin but you would have to make a connector to fit. I actually use it on the M4P now, but I Also have a microprobe so my z-stop was free to use.

Coolbobjoe, 2023-12-18:
> My sincere apologies. I was in another repo and talking with several others I did notice the repo and assumed it was for the SFS2. For the M4P you might be able to use the probe pin but you would [...]

Can you share how your cfg settings for this? I have the exact same setup and need to get that figured out once I finally get canbus over U2C working finally. Was a bit confused on how to accomplish this.

Pneumanifest, 2023-12-18:
I use the z-stop and e-stop 

`[filament_switch_sensor switch_sensor]
switch_pin: PC15 #E0-stop
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament loss detected

[filament_motion_sensor encoder_sensor]
switch_pin: PC2 #Z-stop
detection_length: 2.88
extruder: extruder
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 No motion detected`

Coolbobjoe, 2024-01-02:
> I use the z-stop and e-stop
>
> ```
> [filament_switch_sensor switch_sensor]
> switch_pin: PC15 #E0-stop
> pause_on_runout: False
> runout_gcode:
>   PAUSE # [pause_resume] is required in printer.cfg
>   M117 [...]

Did you have to repin the connectors or anything? I am only getting a solid red light out of mine.

Pneumanifest, 2024-01-03:
No repining needed. The red and blue is in the first driver endstop and the black and green are in the last. 
[image]

Str8Gamin420, 2024-01-19:
hi, i too am only getting a solid red light!

Hilacha2, 2025-03-05:
hola, lograron hacer funcionar el sensor de filamento?

Jecks85, 2025-05-13:
> No repining needed. The red and blue is in the first driver endstop and the black and green are in the last. [image]

Thank you 😊

[image]