bigtreetech/Manta-M4P #32: How to use TFT PI on DSI with CB2

Kostt3d, 2024-09-28:
Good morning
I managed to get the CB2 to work with a TFT PI 70 on manta m8p but impossible on the M4p. the screen remains black. The assembly instructions for the dsi cable on the 2 manuals are reversed on the TFT screen; I guess the one on the Manta M8p is the correct one with the GND on top and the 3v3 on the bottom. Who can help me please?
[image]
[image]
[armbianEnv.txt](https://github.com/user-attachments/files/17174254/armbianEnv.txt)

[system.txt](https://github.com/user-attachments/files/17174260/system.txt)

Kostt3d, 2024-09-28:
problem solved, made the same connection as for the manta m8p and the TFT pi dsi screen works

Hooba97, 2025-02-10:
I have the same parts 
Manta m8p + cb2 + pi tft43 

I changed the files just as the manual described 
But the screen is black and klipper stopped working

ismaelroque, 2025-03-03:
How did you get this to work I have the cb2 manta m8p v2 with the btt tft43 all I get is a lit black screen.

bigtreetech, 2025-05-05:
There are two types of MIPI ribbon cables. An A and a B type. You may need to use the opposite type if it is not working with your current board.